RoboFlamingo/RoboFlamingo #1: [training] lack of `enrich_lang_annotations.json`

Aopolin-Lv, 2023-11-06:
Hi, author. RoboFlamingo is an interesting work so that I want to reimplement it. However, after installing all required package, the code raise an error that lacking of `enrich_lang_annotations.json` in robot_flamingo/data/data.py line 261. I wonder if you forgot to submit this document to the repo, or is there any other solution to solve this bug?

lixinghang12, 2023-11-06:
Thanks for your remind, the `enrich_lang_annotations.json` file has been uploaded, and we update the `load_partial_traj_data` function in robot_flamingo/data/data.py line 1211.

Aopolin-Lv, 2023-11-08:
May I ask how much resources you need for training, such as what kind of calculation card and how many days it takes?

lixinghang12, 2023-11-08:
> May I ask how much resources you need for training, such as what kind of calculation card and how many days it takes?

We train on 8 A100 gpus, 3b model consumes about 15 hours for one epoch, and 9b model consumes about 25 hours for one epoch, the best performance appears at 3/4 epochs

Aopolin-Lv, 2023-11-08:
> > May I ask how much resources you need for training, such as what kind of calculation card and how many days it takes?
> 
> We train on 8 A100 gpus, 3b model consumes about 15 hours for one epoch, and 9b model consumes about 25 hours for one epoch, the best performance appears at 3/4 epochs

Thank you for the response. Could you provide more details about the training settings? For example,  training batch size, learning rate, learning scheduler, etc. By the way, how to evaluate the model performance when training?

lixinghang12, 2023-11-08:
> > > May I ask how much resources you need for training, such as what kind of calculation card and how many days it takes?
> > 
> > 
> > We train on 8 A100 gpus, 3b model consumes about 15 hours for one epoch, and 9b model consumes about 25 hours for one epoch, the best performance appears at 3/4 epochs
> 
> Thank you for the response. I also have following questions:
> 
> 1. Could you provide more details about the training settings? For example,  training batch size, learning rate, learning scheduler, etc.
> 2. By the way, how to evaluate the model performance when training?
> 3. And what's the meaning of the winodw size? From the code, I notice it is set to 12. What happens if the window size is set to a large number, such as 64? I also wonder if a batch data will contain different lang tasks when it forwards.

1. The Training script in the ReadMe is the default setting. The batch size is 48, learning rate is 1e-4 with constant scheduler
2. You may evaluate the imitation loss over unseen demonstrations. However, the success rate is not necessarily positively related to the imitation loss, so in our implementation, we save each the model's parameter at the end of each epoch and evaluate its performance over the 1000 testing episodes.
3. We have tried different window size (8, 12, 16, 24) of LLaMA-based 9b RoboFlamingo, the performance of different window sizes are close while window size=12 achieves the best performance. Intuitively, setting the window size larger would reduce the number of demonstrations seen in each epoch and would requires much more demonstrations to learn a good performance model. However, this idea hasn't been verified yet,  you are more than welcome to try different window size with different VLM backbones.

closed 2023-11-14 by Ericonaldo

Aopolin-Lv, 2023-11-15:
Hi, when evaluating, I found that the enviroment exists bugs.
1. When I install the `hulc` and `calvin` repos, it notifies me that the `torch` version should be 1.13. However, its version in this repo is 1.12.1+cu113.
2. Should I install open flamingo from this repo by using the command like `cd open_flamingo && python setup.py install`?
It seems that`open_flamingo` repo need the `torch==2.0.1`.
So could you provide some suggestions?

lixinghang12, 2023-11-15:
> Hi, when evaluating, I found that the enviroment exists bugs.
> 
> 1. When I install the `hulc` and `calvin` repos, it notifies me that the `torch` version should be 1.13. However, its version in this repo is 1.12.1+cu113.
> 2. Should I install open flamingo from this repo by using the command like `cd open_flamingo && python setup.py install`? It seems that`open_flamingo` repo need the `torch==2.0.1`.
> 3. The python version given by this repo is 3.8, while the open_flamingo requires `python==3.9`
> 
> So could you provide some suggestions?

In our implementation,  `torch==2.0.1` and `python==3.8`, fortunately, the CALVIN Env and open_flamingo still works, you can follow our setting to install the environment.

Aopolin-Lv, 2023-11-15:
> In our implementation, `torch==2.0.1` and `python==3.8`, fortunately, the CALVIN Env and open_flamingo still works, you can follow our setting to install the environment.

Thank you. After installing required packages and `export MESA_GL_VERSION_OVERRIDE = 4.1`, I still come across following bug:
***
```
Flamingo model initialized with 1088187703 trainable parameters. 
Loading origin flamingo checkpoint from  /data/Models/OpenFlamingo-3B-vitl-mpt1b-langinstruct/checkpoint.pt. 
Loading robot-flamingo checkpoint from  RobotFlamingoDBG/checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth
argv[0]=--width=200  
argv[1]=--height=200 
EGL device choice: -1 of 1.
Loaded EGL 1.4 after reload.
GL_VENDOR=VMware, Inc.
GL_RENDERER=llvmpipe (LLVM 10.0.0, 256 bits)
GL_VERSION=4.1 (Compatibility Profile) Mesa 20.0.8
GL_SHADING_LANGUAGE_VERSION=3.30
Version = 4.1 (Compatibility Profile) Mesa 20.0.8
Vendor = VMware, Inc.
Renderer = llvmpipe (LLVM 10.0.0, 256 bits)
Exception in thread Thread-7:
Traceback (most recent call last):
  File "/opt/conda/envs/rob/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/opt/conda/envs/rob/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/root/calvin/calvin_env/tacto/tacto/renderer_mp.py", line 13, in worker
    renderer = _Renderer(width, height, background, config_path)
  File "/root/calvin/calvin_env/tacto/tacto/renderer.py", line 106, in __init__
    self._init_pyrender()
  File "/root/calvin/calvin_env/tacto/tacto/renderer.py", line 146, in _init_pyrender
    colors, depths = self.render(object_poses=None, noise=False, calibration=False)
  File "/root/calvin/calvin_env/tacto/tacto/renderer.py", line 568, in render
    color, depth = self.r.render(self.scene, flags=self.flags_render)
  File "/opt/conda/envs/rob/lib/python3.8/site-packages/pyrender/offscreen.py", line 102, in render
    retval = self._renderer.render(scene, flags, seg_node_map)
  File "/opt/conda/envs/rob/lib/python3.8/site-packages/pyrender/renderer.py", line 144, in render
    retval = self._forward_pass(scene, flags, seg_node_map=seg_node_map)
  File "/opt/conda/envs/rob/lib/python3.8/site-packages/pyrender/renderer.py", line 326, in _forward_pass
    self._configure_forward_pass_viewport(flags)
  File "/opt/conda/envs/rob/lib/python3.8/site-packages/pyrender/renderer.py", line 1012, in _configure_forward_pass_viewport
    self._configure_main_framebuffer()
  File "/opt/conda/envs/rob/lib/python3.8/site-packages/pyrender/renderer.py", line 1092, in _configure_main_framebuffer
    glRenderbufferStorageMultisample(
  File "/opt/conda/envs/rob/lib/python3.8/site-packages/OpenGL/platform/baseplatform.py", line 409, in __call__
    return self( *args, **named )
  File "/opt/conda/envs/rob/lib/python3.8/site-packages/OpenGL/error.py", line 228, in glCheckError
    raise GLError(
OpenGL.error.GLError: GLError(
        err = 1282,
        baseOperation = glRenderbufferStorageMultisample,
        cArguments = (GL_RENDERBUFFER, 4, GL_RGBA, 120, 160)
)
```
***
The gpu is a800 and I dont know how to do the headless setting. Could you give me some advice?

Aopolin-Lv, 2023-11-16:
> > > Hi, when evaluating, I found that the enviroment exists bugs.
> > > 
> > > 1. When I install the `hulc` and `calvin` repos, it notifies me that the `torch` version should be 1.13. However, its version in this repo is 1.12.1+cu113.
> > > 2. Should I install open flamingo from this repo by using the command like `cd open_flamingo && python setup.py install`? It seems that`open_flamingo` repo need the `torch==2.0.1`.
> > > 3. The python version given by this repo is 3.8, while the open_flamingo requires `python==3.9`
> > > 
> > > So could you provide some suggestions?
> > 
> > 
> > In our implementation, `torch==2.0.1` and `python==3.8`, fortunately, the CALVIN Env and open_flamingo still works, you can follow our setting to install the environment.
> 
> Thank you. After installing required packages and `export MESA_GL_VERSION_OVERRIDE = 4.1`, I still come across following bug:
> 
> ```
> Flamingo model initialized with 1088187703 trainable parameters. 
> Loading origin flamingo checkpoint from  /data/Models/OpenFlamingo-3B-vitl-mpt1b-langinstruct/checkpoint.pt. 
> Loading robot-flamingo checkpoint from  RobotFlamingoDBG/checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth
> argv[0]=--width=200  
> argv[1]=--height=200 
> EGL device choice: -1 of 1.
> Loaded EGL 1.4 after reload.
> GL_VENDOR=VMware, Inc.
> GL_RENDERER=llvmpipe (LLVM 10.0.0, 256 bits)
> GL_VERSION=4.1 (Compatibility Profile) Mesa 20.0.8
> GL_SHADING_LANGUAGE_VERSION=3.30
> Version = 4.1 (Compatibility Profile) Mesa 20.0.8
> Vendor = VMware, Inc.
> Renderer = llvmpipe (LLVM 10.0.0, 256 bits)
> Exception in thread Thread-7:
> Traceback (most recent call last):
>   File "/opt/conda/envs/rob/lib/python3.8/threading.py", line 932, in _bootstrap_inner
>     self.run()
>   File "/opt/conda/envs/rob/lib/python3.8/threading.py", line 870, in run
>     self._target(*self._args, **self._kwargs)
>   File "/root/calvin/calvin_env/tacto/tacto/renderer_mp.py", line 13, in worker
>     renderer = _Renderer(width, height, background, config_path)
>   File "/root/calvin/calvin_env/tacto/tacto/renderer.py", line 106, in __init__
>     self._init_pyrender()
>   File "/root/calvin/calvin_env/tacto/tacto/renderer.py", line 146, in _init_pyrender
>     colors, depths = self.render(object_poses=None, noise=False, calibration=False)
>   File "/root/calvin/calvin_env/tacto/tacto/renderer.py", line 568, in render
>     color, depth = self.r.render(self.scene, flags=self.flags_render)
>   File "/opt/conda/envs/rob/lib/python3.8/site-packages/pyrender/offscreen.py", line 102, in render
>     retval = self._renderer.render(scene, flags, seg_node_map)
>   File "/opt/conda/envs/rob/lib/python3.8/site-packages/pyrender/renderer.py", line 144, in render
>     retval = self._forward_pass(scene, flags, seg_node_map=seg_node_map)
>   File "/opt/conda/envs/rob/lib/python3.8/site-packages/pyrender/renderer.py", line 326, in _forward_pass
>     self._configure_forward_pass_viewport(flags)
>   File "/opt/conda/envs/rob/lib/python3.8/site-packages/pyrender/renderer.py", line 1012, in _configure_forward_pass_viewport
>     self._configure_main_framebuffer()
>   File "/opt/conda/envs/rob/lib/python3.8/site-packages/pyrender/renderer.py", line 1092, in _configure_main_framebuffer
>     glRenderbufferStorageMultisample(
>   File "/opt/conda/envs/rob/lib/python3.8/site-packages/OpenGL/platform/baseplatform.py", line 409, in __call__
>     return self( *args, **named )
>   File "/opt/conda/envs/rob/lib/python3.8/site-packages/OpenGL/error.py", line 228, in glCheckError
>     raise GLError(
> OpenGL.error.GLError: GLError(
>         err = 1282,
>         baseOperation = glRenderbufferStorageMultisample,
>         cArguments = (GL_RENDERBUFFER, 4, GL_RGBA, 120, 160)
> )
> ```
> 
> The gpu is a800 and I dont know how to do the headless setting. Could you give me some advice? And what's the version of `PyOpenGL`?`

I don't use docker to excute the program, and the problem is solved.
When creating the object of `PlayTableSimEnv`, it need instantiate all cameras including gripper, static and tactile. However, the tactile camera cant not instantiate by `hydra.utils.instantiate`. Have you come across the problem?

lixinghang12, 2023-11-16:
> I don't use docker to excute the program, and the problem is solved. When creating the object of `PlayTableSimEnv`, it need instantiate all cameras including gripper, static and tactile. However, the tactile camera cant not instantiate by `hydra.utils.instantiate`. Have you come across the problem?

In our setting, the `RoboFlamingo` model only takes the static and gripper camera as input, so we don't need to instantiate the tactile camera.

Aopolin-Lv, 2023-11-16:
> In our setting, the `RoboFlamingo` model only takes the static and gripper camera as input, so we don't need to instantiate the tactile camera.

So in theory, I commented out the tactile part of the configuration file, which didn't affect the final performance, right?