RoboFlamingo/RoboFlamingo #2: Exception in thread Thread-1:

le-wei, 2023-11-09:
Hello,

Thank you very much for open-sourcing such an interesting project. I followed the prompt steps and conducted a test. However, there was an error during the rendering process, which caused the thread to die. The log is as follows:

**[2023-11-09 11:52:33,764] torch.distributed.run: [WARNING] 
[2023-11-09 11:52:33,764] torch.distributed.run: [WARNING] *****************************************
[2023-11-09 11:52:33,764] torch.distributed.run: [WARNING] Setting OMP_NUM_THREADS environment variable for each process to be 1 in default, to avoid your system being overloaded, please further tune the variable for optimal performance in your application as needed. 
[2023-11-09 11:52:33,764] torch.distributed.run: [WARNING] *****************************************
pybullet build time: May 20 2022 19:44:17
pybullet build time: May 20 2022 19:44:17
device_id:  cuda:1
world_size:  2
device_id:  cuda:0
world_size:  2
Using pad_token, but it is not set yet.
/home/ubuntu/objectData/RoboFlamingo/RoboFlamingo/open_flamingo/mpt-1b-redpajama-200b
Explicitly passing a `revision` is encouraged when loading a configuration with custom code to ensure no malicious code has been contributed in a newer revision.
Explicitly passing a `revision` is encouraged when loading a model with custom code to ensure no malicious code has been contributed in a newer revision.
Using pad_token, but it is not set yet.
/home/ubuntu/objectData/RoboFlamingo/RoboFlamingo/open_flamingo/mpt-1b-redpajama-200b
Explicitly passing a `revision` is encouraged when loading a configuration with custom code to ensure no malicious code has been contributed in a newer revision.
Explicitly passing a `revision` is encouraged when loading a model with custom code to ensure no malicious code has been contributed in a newer revision.
WARNING:py.warnings:/home/ubuntu/.cache/huggingface/modules/transformers_modules/mpt-1b-redpajama-200b/attention.py:289: UserWarning: Using `attn_impl: torch`. If your model does not use `alibi` or `prefix_lm` we recommend using `attn_impl: flash` otherwise we recommend using `attn_impl: triton`.
  warnings.warn(

WARNING:py.warnings:/home/ubuntu/.cache/huggingface/modules/transformers_modules/mpt-1b-redpajama-200b/attention.py:289: UserWarning: Using `attn_impl: torch`. If your model does not use `alibi` or `prefix_lm` we recommend using `attn_impl: flash` otherwise we recommend using `attn_impl: triton`.
  warnings.warn(

You are using config.init_device='cpu', but you can also use config.init_device="meta" with Composer + FSDP for fast initialization.
You are using config.init_device='cpu', but you can also use config.init_device="meta" with Composer + FSDP for fast initialization.
window size: 12
1024 2048
MosaicGPTConfig {
  "_name_or_path": "/home/ubuntu/objectData/RoboFlamingo/RoboFlamingo/open_flamingo/mpt-1b-redpajama-200b",
  "alibi": true,
  "alibi_bias_max": 8,
  "architectures": [
    "MosaicGPT"
  ],
  "attn_clip_qkv": null,
  "attn_impl": "torch",
  "attn_pdrop": 0,
  "attn_qk_ln": true,
  "attn_uses_sequence_id": false,
  "auto_map": {
    "AutoConfig": "configuration_mosaic_gpt.MosaicGPTConfig",
    "AutoModelForCausalLM": "mosaic_gpt.MosaicGPT"
  },
  "d_model": 2048,
  "emb_init_std": null,
  "emb_init_uniform_lim": null,
  "emb_pdrop": 0,
  "embedding_fraction": 1.0,
  "fan_mode": "fan_in",
  "init_device": "cpu",
  "init_div_is_residual": true,
  "init_gain": 0,
  "init_nonlinearity": "relu",
  "init_std": 0.02,
  "logit_scale": null,
  "low_precision_layernorm": true,
  "max_seq_len": 2048,
  "mlp_ratio": 4,
  "model_type": "mosaic_gpt",
  "n_heads": 16,
  "n_layers": 24,
  "no_bias": true,
  "param_init_fn": "kaiming_normal_",
  "prefix_lm": false,
  "resid_pdrop": 0,
[... 331 lines cut ...]
torch                    2.1.0
torchaudio               2.1.0
torchdata                0.7.0
torchmetrics             1.2.0
torchtext                0.16.0
torchvision              0.16.0
tqdm                     4.66.1
transformers             4.28.1
trimesh                  4.0.3
triton                   2.1.0
typing_extensions        4.8.0
tzdata                   2023.3
urdfpy                   0.0.22
urllib3                  1.26.18
wandb                    0.15.2
wcwidth                  0.2.9
webdataset               0.2.73
wheel                    0.41.2
yacs                     0.1.8
yarl                     1.9.2
zarr                     2.15.0
zipp                     3.17.0
**

I hope you can take some valuable time to give me some suggestions. Thank you very much.

lixinghang12, 2023-11-09:
It seems like a rendering error in CALVIN environment, you can visualize the rendering results from CALVIN and make sure the rendered observations are correct and start the evaluation. In our local evaluation (with A100 80G and V100 32G GPUs), we install dependencies in `robot_flamingo/pt_eval_ckpts.bash` and `export MESA_GL_VERSION_OVERRIDE=4.1` and solves the rendering problem, you may follow our steps and find the correct `MESA_GL_VERSION_OVERRIDE` version for the RTX4090.

Ericonaldo, 2023-11-21:
Is your problem solved? @le-wei

le-wei, 2023-11-22:
@Ericonaldo 
Sorry, I modified the MESA_GL_VERSION_OVERRIDE parameter but it still didn't render. As I have other work to do, I haven't had time to research how to solve this. If you have a good solution, please share it. Thank you.

Joker-all, 2023-12-06:
I also encountered the same problem, visualization error, I changed the export MESA_ GL_ VERSION_ OverRIDE value, still reporting the same error, GPU using INIDIA TITAN RTX.

Ericonaldo, 2023-12-06:
https://github.com/mees/calvin/issues/11, can check if this works.

I think it may be due to the mismatch between the Pybullet and OpenGL versions. Can you check if these packages' version is matched? Or install the same OpenGL version as shown in `robot_flamingo/pt_eval_ckpts.bash`.

sunchurui, 2024-09-06:
I encountered the same problem, and I haven't found a solution either.

trapoom555, 2024-09-16:
I got it working by deleting `os.environ['PYOPENGL_PLATFORM'] = 'osmesa'` in `robot_flamingo/eval/eval_utils.py`.