RoboFlamingo/RoboFlamingo #3: train_one_epoch_calvin error

HZWHH, 2023-11-21:
Hello,

Thanks for your interesting project.When I run the training code, the error is as follows:

    File "/home/share/anaconda3/envs/robo_py38_/lib/python3.8/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 346, in wrapper
      return f(*args, **kwargs)
    File "robot_flamingo/train/train_calvin.py", line 713, in main
      train_one_epoch_calvin(
    File "/home/share/fhq9wdqo/home/Whtcl-1/whtcl/robotic_arm/RoboFlamingo_mzl/robot_flamingo/train/train_utils.py", line 381, in train_one_epoch_calvin
      for num_steps, batch_calvin in t:
    File "/home/share/anaconda3/envs/robo_py38_/lib/python3.8/site-packages/tqdm/std.py", line 1170, in __iter__
      for obj in iterable:
    File "/home/share/anaconda3/envs/robo_py38_/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 628, in __next__
      data = self._next_data()
    File "/home/share/anaconda3/envs/robo_py38_/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1333, in _next_data
      return self._process_data(data)
    File "/home/share/anaconda3/envs/robo_py38_/lib/python3.8/site-packages/torch/utils/data/dataloader.py", line 1359, in _process_data
      data.reraise()
    File "/home/share/anaconda3/envs/robo_py38_/lib/python3.8/site-packages/torch/_utils.py", line 543, in reraise
      raise exception
  TypeError: Caught TypeError in DataLoader worker process 0.

Could you help me to analyze what is the problem ?

Ericonaldo, 2023-11-21:
what is your torch version?

HZWHH, 2023-11-21:
My torch version is 1.13.1

HZWHH, 2023-11-21:
Batch size is 1 ,set like this:'--batch_size_calvin 1 \'; the GPU is A100*4 and the available memory is about 120G;the dataset is  calvin_debug_dataset.

HZWHH, 2023-11-21:
> > Batch size is 1 ,set like this:'--batch_size_calvin 1 '; the GPU is A100*4 and the available memory is about 120G;the dataset is calvin_debug_dataset.
> 
> Please download the CALVIN Dataset use the calvin_dataset instead.

Thank you, I will try it later.

HZWHH, 2023-11-29:
> > > Batch size is 1 ,set like this:'--batch_size_calvin 1 '; the GPU is A100*4 and the available memory is about 120G;the dataset is calvin_debug_dataset.
> > 
> > 
> > Please download the CALVIN Dataset use the calvin_dataset instead.
> 
> Thank you, I will try it later.

HZWHH, 2023-11-29:
It work, thank you。But there is a new error when training with multiple GPUs. The error is as follows：

ERROR:torch.distributed.elastic.multiprocessing.api:failed (exitcode: -9) local_rank: 0 (pid: 2891467) of binary: anaconda3/envs/robo_py38_1/bin/python3.8
Traceback (most recent call last):
  File "anaconda3/envs/robo_py38_1/bin/torchrun", line 8, in <module>
    sys.exit(main())
  File "anaconda3/envs/robo_py38_1/lib/python3.8/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 346, in wrapper
    return f(*args, **kwargs)
  File "anaconda3/envs/robo_py38_1/lib/python3.8/site-packages/torch/distributed/run.py", line 762, in main
    run(args)
  File "anaconda3/envs/robo_py38_1/lib/python3.8/site-packages/torch/distributed/run.py", line 753, in run
    elastic_launch(
  File "anaconda3/envs/robo_py38_1/lib/python3.8/site-packages/torch/distributed/launcher/api.py", line 132, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "anaconda3/envs/robo_py38_1/lib/python3.8/site-packages/torch/distributed/launcher/api.py", line 246, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
========================================================
robot_flamingo/train/train_calvin.py FAILED
--------------------------------------------------------

Do you know what the problem is？

Ericonaldo, 2023-12-03:
@HZWHH Have you solved your problem?

HZWHH, 2023-12-04:
> @HZWHH Have you solved your problem?

yes ,it was something worng with the computer,thank you.

closed 2023-12-04 by Ericonaldo