RoboFlamingo/RoboFlamingo #6: evaluate error

HZWHH, 2023-12-12:
When I run the test code, the following error occurred in the log.

pybullet build time: Nov 28 2023 23:51:11
args.local_rank:  0
args.rank:  0
args.world_size:  1
device_id:  cuda:0
world_size:  1
../model/mpt-1b-redpajama-200b
WARNING:py.warnings:/root/.cache/huggingface/modules/transformers_modules/mpt-1b-redpajama-200b/attention.py:289: UserWarning: Using `attn_impl: torch`. If your model does not use `alibi` or `prefix_lm` we recommend using `attn_impl: flash` otherwise we recommend using `attn_impl: triton`.
  warnings.warn(

You are using config.init_device='cpu', but you can also use config.init_device="meta" with Composer + FSDP for fast initialization.
window size: 12
1024 2048
MosaicGPTConfig {
  "_name_or_path": "../model/mpt-1b-redpajama-200b",
  "alibi": true,
  "alibi_bias_max": 8,
  "architectures": [
    "MosaicGPT"
  ],
  "attn_clip_qkv": null,
  "attn_impl": "torch",
  "attn_pdrop": 0,
  "attn_qk_ln": true,
  "attn_uses_sequence_id": false,
  "auto_map": {
    "AutoConfig": "configuration_mosaic_gpt.MosaicGPTConfig",
    "AutoModelForCausalLM": "mosaic_gpt.MosaicGPT"
  },
  "d_model": 2048,
  "emb_init_std": null,
  "emb_init_uniform_lim": null,
  "emb_pdrop": 0,
  "embedding_fraction": 1.0,
  "fan_mode": "fan_in",
  "init_device": "cpu",
  "init_div_is_residual": true,
  "init_gain": 0,
  "init_nonlinearity": "relu",
  "init_std": 0.02,
  "logit_scale": null,
  "low_precision_layernorm": true,
  "max_seq_len": 2048,
  "mlp_ratio": 4,
  "model_type": "mosaic_gpt",
  "n_heads": 16,
  "n_layers": 24,
  "no_bias": true,
  "param_init_fn": "kaiming_normal_",
  "prefix_lm": false,
  "resid_pdrop": 0,
  "softmax_scale": null,
  "tokenizer_name": "EleutherAI/gpt-neox-20b",
  "torch_dtype": "float32",
  "transformers_version": "4.36.0",
  "use_cache": false,
  "verbose": 0,
  "vocab_size": 50280
}

----------------------------------------------------------------------------------------------------
transformer.blocks
Flamingo model initialized with 1088187703 trainable parameters
Loading origin flamingo checkpoint from  ../model/OpenFlamingo-3B-vitl-mpt1b/checkpoint.pt
Loading robot-flamingo checkpoint from ../model/checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_3b_2.pth
WARNING:py.warnings:/data/mzl_1115/RoboFlamingo_mzl_test/calvin_env/envs/play_table_env.py:287: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  hydra.initialize(".")

argv[0]=--width=200
argv[1]=--height=200
EGL device choice: -1 of 9.
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
libEGL warning: Not allowed to force software rendering when API explicitly selects a hardware device.
libEGL warning: DRI2: failed to create dri screen
Loaded EGL 1.4 after reload.
Unable to create EGL context (eglError: 12292)
ERROR:torch.distributed.elastic.multiprocessing.api:failed (exitcode: 1) local_rank: 0 (pid: 87261) of binary: /root/anaconda3/envs/robo_py38/bin/python
Traceback (most recent call last):
  File "/root/anaconda3/envs/robo_py38/bin/torchrun", line 8, in <module>
    sys.exit(main())
  File "/root/anaconda3/envs/robo_py38/lib/python3.8/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 346, in wrapper
    return f(*args, **kwargs)
  File "/root/anaconda3/envs/robo_py38/lib/python3.8/site-packages/torch/distributed/run.py", line 762, in main
    run(args)
  File "/root/anaconda3/envs/robo_py38/lib/python3.8/site-packages/torch/distributed/run.py", line 753, in run
    elastic_launch(
  File "/root/anaconda3/envs/robo_py38/lib/python3.8/site-packages/torch/distributed/launcher/api.py", line 132, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/root/anaconda3/envs/robo_py38/lib/python3.8/site-packages/torch/distributed/launcher/api.py", line 246, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
robot_flamingo/eval/eval_calvin.py FAILED
------------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
------------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2023-12-12_15:56:28
  host      : A100-8-test-01
  rank      : 0 (local_rank: 0)
  exitcode  : 1 (pid: 87261)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
============================================================

The gpu is also A100 80G. Do you know what caused this error?thank you.

lixinghang12, 2023-12-13:
@HZWHH I think this might be an EGL rendering error. Could you set os.environ['CUDA_LAUNCH_BLOCKING'] = "1" and rerun the `robot_flamingo/pt_eval_ckpts.bash` to get a more detailed log?

HZWHH, 2023-12-13:
> @HZWHH I think this might be an EGL rendering error. Could you set os.environ['CUDA_LAUNCH_BLOCKING'] = "1" and rerun the `robot_flamingo/pt_eval_ckpts.bash` to get a more detailed log?

I have already set os.environ['CUDA_LAUNCH_BLOCKING'] = "1" in the "eval_calvin.py" , is it right ?

Ericonaldo, 2023-12-13:
Seems like a similar problem as mentioned in [https://github.com/RoboFlamingo/RoboFlamingo/issues/2](url) ? Can you try to  verify the version of OpenGL and pybullet?

HZWHH, 2023-12-18:
> Seems like a similar problem as mentioned in [https://github.com/RoboFlamingo/RoboFlamingo/issues/2](url) ? Can you try to verify the version of OpenGL and pybullet?

OpenGL is 3.1.0 and pybullet is 3.2.6。so set MESA_GL_VERSION_OVERRIDE=3.1?  It is still the same error

lixinghang12, 2023-12-19:
@HZWHH Could you try to install the same version of OpenGL & MESA_GL_VERSION_OVERRIDE (4.1), pybullet==3.2.5 and pyrender==0.1.45 and other dependencies following `robot_flamingo/pt_eval_ckpts.bash`? If there is still error, please set os.environ['CUDA_LAUNCH_BLOCKING'] = "1" and update the error log.

closed 2024-01-17 by lixinghang12