RoboFlamingo/RoboFlamingo #7: python version error

mnm117, 2024-01-05:
Hello, as mentioned I used Python 3.8 for running the code, But I faced the below error when I wanted to run " pip install -r requirements.txt "
```

ERROR: Ignored the following versions that require a different python version: 0.22.0 Requires-Python >=3.9; 0.22.0rc1 Requires-Python >=3.9; 1.11.0 Requires-Python <3.13,>=3.9; 1.11.0rc1 Requires-Python <3.13,>=3.9; 1.11.0rc2 Requires-Python <3.13,>=3.9; 1.11.1 Requires-Python <3.13,>=3.9; 1.11.2 Requires-Python <3.13,>=3.9; 1.11.3 Requires-Python <3.13,>=3.9; 1.11.4 Requires-Python >=3.9; 1.12.0rc1 Requires-Python >=3.9; 1.4.0rc1 Requires-Python >=3.9
ERROR: Could not find a version that satisfies the requirement orch==1.12.1+cu113 (from versions: none)
ERROR: No matching distribution found for orch==1.12.1+cu113

```

Ericonaldo, 2024-01-05:
Seems it is a typo and we now fix the requirement.txt. It should be `torch`

closed 2024-01-17 by lixinghang12

emrecanacikgoz, 2024-01-19:
Is it a must to use python 3.8 (e.g. for calvin framework and etc.)? As mentioned by @mnm117, 3.8 results with a lot of conflicts. My case: 
```
git clone https://github.com/RoboFlamingo/RoboFlamingo.git
conda create -n RoboFlamingo python=3.8 -y
source activate RoboFlamingo
pip install -r requirements.txt
```
Errors:
```
ERROR: Ignored the following versions that require a different python version: 0.22.0 Requires-Python >=3.9; 0.22.0rc1 Requires-Python >=3.9; 1.11.0 Requires-Python <3.13,>=3.9; 1.11.0rc1 Requires-Python <3.13,>=3.9; 1.11.0rc2 Requires-Python <3.13,>=3.9; 1.11.1 Requires-Python <3.13,>=3.9; 1.11.2 Requires-Python <3.13,>=3.9; 1.11.3 Requires-Python <3.13,>=3.9; 1.11.4 Requires-Python >=3.9; 1.12.0rc1 Requires-Python >=3.9; 1.12.0rc2 Requires-Python >=3.9; 1.25.0 Requires-Python >=3.9; 1.25.0rc1 Requires-Python >=3.9; 1.25.1 Requires-Python >=3.9; 1.25.2 Requires-Python >=3.9; 1.26.0 Requires-Python <3.13,>=3.9; 1.26.0b1 Requires-Python <3.13,>=3.9; 1.26.0rc1 Requires-Python <3.13,>=3.9; 1.26.1 Requires-Python <3.13,>=3.9; 1.26.2 Requires-Python >=3.9; 1.26.3 Requires-Python >=3.9; 1.4.0 Requires-Python >=3.9; 1.4.0rc1 Requires-Python >=3.9; 2.1.0 Requires-Python >=3.9; 2.1.0rc0 Requires-Python >=3.9; 2.1.1 Requires-Python >=3.9; 2.1.2 Requires-Python >=3.9; 2.1.3 Requires-Python >=3.9; 2.1.4 Requires-Python >=3.9; 2.2.0rc0 Requires-Python >=3.9
ERROR: Could not find a version that satisfies the requirement torch==1.12.1+cu113 (from versions: 1.4.0, 1.5.0, 1.5.1, 1.6.0, 1.7.0, 1.7.1, 1.8.0, 1.8.1, 1.9.0, 1.9.1, 1.10.0, 1.10.1, 1.10.2, 1.11.0, 1.12.0, 1.12.1, 1.13.0, 1.13.1, 2.0.0, 2.0.1, 2.1.0, 2.1.1, 2.1.2)
ERROR: No matching distribution found for torch==1.12.1+cu113
```

lixinghang12, 2024-01-21:
@emrecanacikgoz You can use new version like python 3.10. For the pytorch installation error, you can delete the '+cu113' suffix and try again.