RoboFlamingo/RoboFlamingo #8: Training result issues

YiyangHuang-work, 2024-01-07:
Thank you for your work. Can you give me the specific parameters of your training? I used ABCD data to train according to the current parameters on github. The loss is equivalent to the output.log you published, but the success rate during testing is compared to the results you published. The difference is about 15%. I want to confirm what is wrong with my training.

lixinghang12, 2024-01-08:
> Thank you for your work. Can you give me the specific parameters of your training? I used ABCD data to train according to the current parameters on github. The loss is equivalent to the output.log [...]

We have added the training bash [pt_run_gripper_post_ws_12_traj_aug_mpt_dolly_3b.bash](https://github.com/RoboFlamingo/RoboFlamingo/blob/main/robot_flamingo/pt_run_gripper_post_ws_12_traj_aug_mpt_dolly_3b.bash) in our repo, it contains all the hyper parameters in our implementation. Specifically, we trained our model on 8 A100 GPUs with 6*8 batch size, if you have modified the batch size, you should also change the learning rate.

YiyangHuang-work, 2024-01-08:
Thank you for your reply. I used the same 8*A100 and batch size 6 as you, this is my [output.log](https://github.com/RoboFlamingo/RoboFlamingo/files/13859509/output.log)
I saw that the loss did not reach 2e-3 in the output.log you provided, so I trained for 5 more epochs. This is my test result
[image]
besides,here are the results for each type of task in my test epoch 5 and 10:[result.log](https://github.com/RoboFlamingo/RoboFlamingo/files/13859811/result.log)
[anas-awadalla/mpt-1b-redpajama-200b-dolly](https://huggingface.co/openflamingo/OpenFlamingo-3B-vitl-mpt1b-langinstruct)，I downloaded the RoboFlamingo model from here, and it worked best in your experiment. The rest of the parameter settings remain unchanged from your github. I also tried the training of ABC->D, and the success rate was about 20% lower than the results in the paper.Could you please check if there are any abnormalities in my experimental data? Thank you very much~

lixinghang12, 2024-01-09:
@YiyangHuang-work Could you provide me full launch code for your training? Make sure that you have correctly downloaded the ViT-L, MPT and OpenFlamingo Parameters and loaded them before finetuning the model. Also, we have provided our pre-trained model on HuggingFace, you can download the mpt-3b-dolly version and evaluate on your local machine to make sure that the evaluation process is correct.

YiyangHuang-work, 2024-01-09:
cofinetune_checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_3.pth
[image]

YiyangHuang-work, 2024-01-09:
I tested the two checkpoints you announced, and the success rates were both around 80%. After comparison, I thought this was the result of ABC->D training?
cofinetune_checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_3.pth evaluation result is:
[image]
This is the bash I use for training:[train.txt](https://github.com/RoboFlamingo/RoboFlamingo/files/13869821/train.txt)
This is the log of the beginning of training:[train.log](https://github.com/RoboFlamingo/RoboFlamingo/files/13869903/train.log)

lixinghang12, 2024-01-09:
@YiyangHuang-work Can you provide the evaluation bash and test the checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth? And make sure that you have replaced the pretrained model paths in mpt_dict in [factory.py](https://github.com/RoboFlamingo/RoboFlamingo/blob/main/robot_flamingo/models/factory.py)

YiyangHuang-work, 2024-01-09:
This is the evaluation bash:[evaluation.txt](https://github.com/RoboFlamingo/RoboFlamingo/files/13871044/evaluation.txt)
I eval the checkpoint checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth for 100 sequences,the result is like this:[evaluation.log](https://github.com/RoboFlamingo/RoboFlamingo/files/13871097/evaluation.log)
I replaced the pretrained mdel paths in mpt_dict:
[image]
What should your checkpoint evaluation results be? Thank you for your reply

YiyangHuang-work, 2024-01-11:
@lixinghang12 
I evaluated the checkpoint you submitted on 1,000 sequences. This is my evaluation result[evaluation.txt](https://github.com/RoboFlamingo/RoboFlamingo/files/13896884/evaluation.txt). Could you tell me the checkpoint you published trained on the ABCD data set or the ABC data set？I did not find the corresponding value in the article,I want to confirm if there is a problem with my testing process,Thank you very much.

lixinghang12, 2024-01-11:
@YiyangHuang-work Thanks for your feedback, we would download the pre-trained model from Hugging Face and evaluate it on our local machine to locate the problem. You can download other models and check them if you have time. Also, you can visualize the success and failed episodes to verify whether the evaluation env and process is correct.

YiyangHuang-work, 2024-01-11:
@lixinghang12 I re-checked the [Openflamingo](https://huggingface.co/openflamingo/OpenFlamingo-3B-vitl-mpt1b-langinstruct) checkpoint I used and I could not find the problem. In addition, I trained for a few more epochs based on the [checkpoint](https://huggingface.co/roboflamingo/RoboFlamingo/blob/main/checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth) you published and found that the evaluation results were basically the same as the [evaluation.txt](https://github.com/RoboFlamingo/RoboFlamingo/files/13896884/evaluation.txt). This may indicate that there is not much problem with my training process. I also train the model on ABC dataset several times, and test on D. In my multiple trainings, there is a 10% difference in the task success rate of checkpoints with the same loss. the highest success rate of task1 is around 58%. So if possible, It's really kind of you to retry the training process of the repo on ABC dataset, or give some empirical parameters for stable training,I still couldn't find the problem in the fine-tuning process, thank you again for your reply.

YiyangHuang-work, 2024-01-23:
I found the problem, I'm not using GPU rendering, thanks for the help

emrecanacikgoz, 2024-02-18:
Hi @YiyangHuang-work, how did you fix GPU rendering issue?

In more details, how did you first understand that you are not using GPU rendering and then how exactly you fixed it?