RoboFlamingo/RoboFlamingo #9: Release code of co-train？

fengxiuyaun, 2024-01-08:


lixinghang12, 2024-01-08:
The co-train code has been updated, please see `[pt_run_gripper_post_ws_12_traj_aug_mpt_dolly_3b_co_train.bash](https://github.com/RoboFlamingo/RoboFlamingo/blob/main/robot_flamingo/pt_run_gripper_post_ws_12_traj_aug_mpt_dolly_3b_co_train.bash)` for more details.

fengxiuyaun, 2024-01-09:
Yes, but  "cotrain" is not used

lixinghang12, 2024-01-10:
@fengxiuyaun Sorry, there's a typeo in line 21 of [pt_run_gripper_post_ws_12_traj_aug_mpt_dolly_3b_co_train.bash](https://github.com/RoboFlamingo/RoboFlamingo/blob/main/robot_flamingo/pt_run_gripper_post_ws_12_traj_aug_mpt_dolly_3b_co_train.bash), i have fix it from `co_train` to `cotrain`.

COST-97, 2024-03-03:
Hello:
The code for cotrain seems to be incomplete.
1. In train_calvin.py, `train_one_epoch_calvin_cotrain` is not called.
2. The `output.logits` in the flamingo_mpt.py are only the corresponding `lm_head` of the robot tasks, which should not be compatible with coco and VQA data, right?
Could you add to this part of the code?
Thank you!