RoboFlamingo/RoboFlamingo #11: pip install -r requirements.txt

zydczqcnydx, 2024-01-29:
python=3.8
[image]
python=3.9
[image]

May I ask how to solve it？

sun123-cmd, 2024-01-29:
Maybe you need to delete some modules of txt list, and install them independly

jyue86, 2024-02-05:
Stick with the `python=3.8` conda env. I think a lot of the problems arise from not using pytorch cu+11.3, which I resolved with:
```
pip install  -r requirements.txt --extra-index-url https://download.pytorch.org/whl/cu113
```

closed 2024-04-12 by Ericonaldo