RoboFlamingo/RoboFlamingo #13: No module named 'robot_flamingo'

emrecanacikgoz, 2024-02-02:
Hi,

When I follow the provided instructions to set the repo and try for training, I am facing the error below:

```
WARNING:torch.distributed.run:
*****************************************
Setting OMP_NUM_THREADS environment variable for each process to be 1 in default, to avoid your system being overloaded, please further tune the variable for optimal performance in your application as needed. 
*****************************************
Traceback (most recent call last):
Traceback (most recent call last):
  File "robot_flamingo/train/train_calvin.py", line 16, in <module>
    from robot_flamingo.data.data import get_data
  File "robot_flamingo/train/train_calvin.py", line 16, in <module>
Traceback (most recent call last):
ModuleNotFoundError: No module named 'robot_flamingo'
  File "robot_flamingo/train/train_calvin.py", line 16, in <module>
    from robot_flamingo.data.data import get_data
ModuleNotFoundError: No module named 'robot_flamingo'
    from robot_flamingo.data.data import get_data
ModuleNotFoundError: No module named 'robot_flamingo'
ERROR:torch.distributed.elastic.multiprocessing.api:failed (exitcode: 1) local_rank: 0 (pid: 33566) of binary: /kuacc/users/eacikgoz17/.conda/envs/RoboFlamingo/bin/python
Traceback (most recent call last):
  File "/kuacc/users/eacikgoz17/.conda/envs/RoboFlamingo/bin/torchrun", line 8, in <module>
    sys.exit(main())
  File "/kuacc/users/eacikgoz17/.conda/envs/RoboFlamingo/lib/python3.8/site-packages/torch/distributed/elastic/multiprocessing/errors/__init__.py", line 346, in wrapper
    return f(*args, **kwargs)
  File "/kuacc/users/eacikgoz17/.conda/envs/RoboFlamingo/lib/python3.8/site-packages/torch/distributed/run.py", line 762, in main
    run(args)
  File "/kuacc/users/eacikgoz17/.conda/envs/RoboFlamingo/lib/python3.8/site-packages/torch/distributed/run.py", line 753, in run
    elastic_launch(
  File "/kuacc/users/eacikgoz17/.conda/envs/RoboFlamingo/lib/python3.8/site-packages/torch/distributed/launcher/api.py", line 132, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/kuacc/users/eacikgoz17/.conda/envs/RoboFlamingo/lib/python3.8/site-packages/torch/distributed/launcher/api.py", line 246, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
robot_flamingo/train/train_calvin.py FAILED
------------------------------------------------------------
```

Below is the environment info:
```
(RoboFlamingo) [eacikgoz17@login03 RoboFlamingo]$ conda list
# packages in environment at /kuacc/users/eacikgoz17/.conda/envs/RoboFlamingo:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
aiohttp                   3.9.3                    pypi_0    pypi
aiosignal                 1.3.1                    pypi_0    pypi
antlr4-python3-runtime    4.8                      pypi_0    pypi
appdirs                   1.4.4                    pypi_0    pypi
asciitree                 0.3.3                    pypi_0    pypi
async-timeout             4.0.3                    pypi_0    pypi
attrs                     23.2.0                   pypi_0    pypi
ca-certificates           2023.12.12           h06a4308_0  
calvin                    0.0.1                     dev_0    <develop>
calvin-env                0.0.1                    pypi_0    pypi
certifi                   2023.11.17               pypi_0    pypi
cffi                      1.16.0                   pypi_0    pypi
charset-normalizer        3.3.2                    pypi_0    pypi
click                     8.1.7                    pypi_0    pypi
cloudpickle               3.0.0                    pypi_0    pypi
cmake                     3.18.4                   pypi_0    pypi
colorlog                  6.8.2                    pypi_0    pypi
contourpy                 1.1.1                    pypi_0    pypi
cycler                    0.12.1                   pypi_0    pypi
decorator                 4.4.2                    pypi_0    pypi
docker-pycreds            0.4.0                    pypi_0    pypi
[... 164 lines cut ...]
    --report_to_wandb \
    --llm_name mpt_dolly_3b \
    --traj_cons \
    --use_gripper \
    --fusion_mode post \
    --rgb_pad 10 \
    --gripper_pad 4 \
    --precision fp32 \
    --num_epochs 5 \
    --gradient_accumulation_steps 1 \
    --batch_size_calvin 6 \
    --run_name RobotFlamingoDBG \
    --calvin_dataset /datasets/calvin_new/task_ABCD_D \
    --openflamingo_checkpoint RoboFlamingo/RobotFlamingoDBG/checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_3b_4.pth \
    --cross_attn_every_n_layers 4 \
    --dataset_resampled \
    --loss_multiplier_calvin 1.0 \
    --workers 1 \
    --lr_scheduler constant \
    --warmup_steps 5000 \
    --learning_rate 1e-4 \
    --save_every_iter 10000 \
    --from_scratch \
```
Could you please help me for that path issue?

Ericonaldo, 2024-02-07:
Hi. Are you running the command from the parent directory of the `robot_flamingo` directory?

If so and it still not work, you can try to add the path into the python sys.path, in your bash or add a code to the training script:
```
import sys
sys.path.append('/path/to/the/directory')
```

emrecanacikgoz, 2024-02-14:
Issue solved, thank you.

closed 2024-02-14 by emrecanacikgoz