RoboFlamingo/RoboFlamingo #15: can this method use in real env directly?

lunalulu, 2024-02-20:
thanks for your great job~

Ericonaldo, 2024-02-20:
We have not released the version that is fine-tuned on real robot data. If you have such data, you can try to fine-tune and then deploy on your setup.

lunalulu, 2024-02-20:
> We have not released the version that is fine-tuned on real robot data. If you have such data, you can try to fine-tune and then deploy on your setup.

thanks 
When will the real robot data training version be released?

Ericonaldo, 2024-02-20:
We are making efforts to do that :) We also want it to be released as soon as possible.

closed 2024-02-23 by Ericonaldo

lunalulu, 2024-03-05:
> We are making efforts to do that :) We also want it to be released as soon as possible.

thanks very much~
please, How to collect real robot data? Can you give me some advice? For example, how to set the collection frequency? some scripts?

Ericonaldo, 2024-03-05:
Actually I recommend you to refer to existing published works about that :)

lunalulu, 2024-03-05:
> Actually I recommend you to refer to existing published works about that :)

for example?

lunalulu, 2024-04-11:
> Actually I recommend you to refer to existing published works about that :)

Can the following real collection data be made open?
thanks very much

data_dir="/mnt/bn/robotics-data-hl/real_data/mode1_data_pick_place_001_1023/"

lunalulu, 2024-04-24:
@Ericonaldo What is the approximate minimum number of trajectories per task that is suitable when collecting real-world operational data for fine-tuning a model?
thanks very much~