RoboFlamingo/RoboFlamingo #19: Evaluation on machine without sudo

dingjiansw101, 2024-02-23:
Dear Authors,

Thanks for sharing code for this excellent work. I tried to use "pt_eval_ckpt.bash" for the evaluation. But I did not have the sudo. How can I install all the required libraries?

Best,
Jian Ding

Ericonaldo, 2024-02-23:
Can you use docker?

dingjiansw101, 2024-02-23:
I can use singularity. Could you provide me with an image?

Ericonaldo, 2024-02-26:
We use an internal docker system so I can only show you the following dockerfile, note that the base image should be an image that can run pytorch programs.
```
FROM /path/to/a/pytorch/base/image

# 1 apt 依赖
RUN apt-get update \
    && apt-get install -yq --no-install-recommends \
        build-essential cmake pkg-config autoconf libtool automake \
        vim git tree wget curl ca-certificates unzip \
        iputils-ping net-tools ffmpeg libsm6 libxext6 libgl1 python3-opencv \
        libegl1-mesa libegl1 libosmesa6-dev patchelf \
    && rm -rf /var/lib/apt/lists/*

RUN python3 -m pip install -U pip && python3 -m pip uninstall -y byted-wandb
RUN python3 -m pip install --no-cache-dir \
    opencv-python==4.7.0.72 opencv-python-headless==4.7.0.72 pytorch_lightning \
    python-dotenv tqdm jupyter-client==7.4.4 numpy==1.19.5 g2p_en \
    einops einops-exts pillow more-itertools \
    braceexpand webdataset nltk scipy inflection sentencepiece open_clip_torch \
    black mypy pylint pytest requests tensorboard transformers datasets
RUN python3 -m pip install --no-cache-dir pytest-benchmark pytest-runner setuptools==57.5.0
RUN python3 -m pip install --no-cache-dir mmcv -f https://download.openmmlab.com/mmcv/dist/cu117/torch2.0/index.html
RUN python3 -m pip install pyhash
RUN python3 -m pip install open-flamingo

RUN sed -i "s/^from decimal import Decimal/from math import gcd\n&/" /usr/lib/python3.9/fractions.py
RUN python3 -m pip uninstall -y wandb && python3 -m pip install -U --force-reinstall wandb

RUN rm /usr/bin/python && ln -s /usr/bin/python3 /usr/bin/python
```

dingjiansw101, 2024-02-26:
Thanks for providing this. 
I created a docker several days before. I can run on the evaluation on the local machine. However, I met the error on the slurm machines.

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/opt/conda/envs/RoboFlamingo/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/opt/conda/envs/RoboFlamingo/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/ibex/user/dingj0b/code/calvin/calvin_env/tacto/tacto/renderer_mp.py", line 13, in worker
    renderer = _Renderer(width, height, background, config_path)
  File "/ibex/user/dingj0b/code/calvin/calvin_env/tacto/tacto/renderer.py", line 106, in __init__
    self._init_pyrender()
  File "/ibex/user/dingj0b/code/calvin/calvin_env/tacto/tacto/renderer.py", line 144, in _init_pyrender
    self.r = pyrender.OffscreenRenderer(self.width, self.height)
  File "/opt/conda/envs/RoboFlamingo/lib/python3.8/site-packages/pyrender/offscreen.py", line 31, in __init__
    self._create()
  File "/opt/conda/envs/RoboFlamingo/lib/python3.8/site-packages/pyrender/offscreen.py", line 149, in _create
    self._platform.init_context()
  File "/opt/conda/envs/RoboFlamingo/lib/python3.8/site-packages/pyrender/platforms/egl.py", line 177, in init_context
    assert eglInitialize(self._egl_display, major, minor)
  File "/home/dingj0b/.local/lib/python3.8/site-packages/OpenGL/platform/baseplatform.py", line 409, in __call__
    return self( *args, **named )
  File "/home/dingj0b/.local/lib/python3.8/site-packages/OpenGL/error.py", line 228, in glCheckError
    raise GLError(
OpenGL.error.GLError: GLError(
        err = 12289,
        baseOperation = eglInitialize,
        cArguments = (
                <OpenGL._opaque.EGLDisplay_pointer object at 0x1481840b2640>,
                c_long(0),
                c_long(0),
        ),
        result = 0
)
libEGL warning: failed to open /dev/dri/renderD135: Permission denied

libEGL warning: failed to open /dev/dri/card7: Permission denied

Have you met similar errors before?

dingjiansw101, 2024-02-27:
> Actually not at all.. 

Did you run experiments on slurm cluster?

Ericonaldo, 2024-02-27:
Not exactly (as we use company resources), but I think the job scheduler may be similar things. Is there any GPU driver installed on this machine? Can you ask the administrator to help you with that?

dingjiansw101, 2024-02-27:
> Not exactly (as we use company resources), but I think the job scheduler may be similar things essentially. Is there any GPU driver installed on this machine? Can you ask the administrator to help you with that?

Thanks for the reply. So you used computer cluster, but not in slurm system, right?

dingjiansw101, 2024-02-28:
> We use an internal docker system so I can only show you the following dockerfile for your reference, note that the base image should be an image that can run pytorch programs.
> 
> ```
> FROM /path/to/a/pytorch/base/image
> 
> 
> RUN apt-get update \
>     && apt-get install -yq --no-install-recommends \
>         build-essential cmake pkg-config autoconf libtool automake \
>         vim git tree wget curl ca-certificates unzip \
>         iputils-ping net-tools ffmpeg libsm6 libxext6 libgl1 python3-opencv \
>         libegl1-mesa libegl1 libosmesa6-dev patchelf \
>     && rm -rf /var/lib/apt/lists/*
> 
> RUN python3 -m pip install -U pip && python3 -m pip uninstall -y byted-wandb
> RUN python3 -m pip install --no-cache-dir \
>     opencv-python==4.7.0.72 opencv-python-headless==4.7.0.72 pytorch_lightning \
>     python-dotenv tqdm jupyter-client==7.4.4 numpy==1.19.5 g2p_en \
>     einops einops-exts pillow more-itertools \
>     braceexpand webdataset nltk scipy inflection sentencepiece open_clip_torch \
>     black mypy pylint pytest requests tensorboard transformers datasets
> RUN python3 -m pip install --no-cache-dir pytest-benchmark pytest-runner setuptools==57.5.0
> RUN python3 -m pip install --no-cache-dir mmcv -f https://download.openmmlab.com/mmcv/dist/cu117/torch2.0/index.html
> RUN python3 -m pip install pyhash
> RUN python3 -m pip install open-flamingo
> 
> RUN sed -i "s/^from decimal import Decimal/from math import gcd\n&/" /usr/lib/python3.9/fractions.py
> RUN python3 -m pip uninstall -y wandb && python3 -m pip install -U --force-reinstall wandb
> 
> RUN rm /usr/bin/python && ln -s /usr/bin/python3 /usr/bin/python
> ```

Hi, why the versions of python library is different from the 
`https://github.com/RoboFlamingo/RoboFlamingo/blob/main/requirements.txt`

Which one should I follow?

dingjiansw101, 2024-02-28:
The previous errors have been solved. Now I have a new problem:

Traceback (most recent call last):
  File "/home/x_dingj/anaconda3/envs/RoboFlamingo/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/home/x_dingj/anaconda3/envs/RoboFlamingo/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/x_dingj/code/calvin/calvin_env/tacto/tacto/renderer_mp.py", line 13, in worker
    renderer = _Renderer(width, height, background, config_path)
  File "/home/x_dingj/code/calvin/calvin_env/tacto/tacto/renderer.py", line 106, in __init__
    self._init_pyrender()
  File "/home/x_dingj/code/calvin/calvin_env/tacto/tacto/renderer.py", line 146, in _init_pyrender
    colors, depths = self.render(object_poses=None, noise=False, calibration=False)
  File "/home/x_dingj/code/calvin/calvin_env/tacto/tacto/renderer.py", line 568, in render
    color, depth = self.r.render(self.scene, flags=self.flags_render)
  File "/home/x_dingj/anaconda3/envs/RoboFlamingo/lib/python3.8/site-packages/pyrender/offscreen.py", line 102, in render
    retval = self._renderer.render(scene, flags, seg_node_map)
  File "/home/x_dingj/anaconda3/envs/RoboFlamingo/lib/python3.8/site-packages/pyrender/renderer.py", line 144, in render
    retval = self._forward_pass(scene, flags, seg_node_map=seg_node_map)
  File "/home/x_dingj/anaconda3/envs/RoboFlamingo/lib/python3.8/site-packages/pyrender/renderer.py", line 369, in _forward_pass
    program = self._get_primitive_program(
  File "/home/x_dingj/anaconda3/envs/RoboFlamingo/lib/python3.8/site-packages/pyrender/renderer.py", line 1000, in _get_primitive_program
    program._add_to_context()
  File "/home/x_dingj/anaconda3/envs/RoboFlamingo/lib/python3.8/site-packages/pyrender/shader_program.py", line 137, in _add_to_context
    self._program_id = gl_shader_utils.compileProgram(*shader_ids)
  File "/home/x_dingj/anaconda3/envs/RoboFlamingo/lib/python3.8/site-packages/OpenGL/GL/shaders.py", line 206, in compileProgram
    program.check_validate()
  File "/home/x_dingj/anaconda3/envs/RoboFlamingo/lib/python3.8/site-packages/OpenGL/GL/shaders.py", line 106, in check_validate
    raise RuntimeError(
RuntimeError: Validation failure (0):

Ericonaldo, 2024-02-29:
> > We use an internal docker system so I can only show you the following dockerfile for your reference, note that the base image should be an image that can run pytorch programs.
> > ```
> > FROM /path/to/a/pytorch/base/image
> > 
> > 
> > RUN apt-get update \
> >     && apt-get install -yq --no-install-recommends \
> >         build-essential cmake pkg-config autoconf libtool automake \
> >         vim git tree wget curl ca-certificates unzip \
> >         iputils-ping net-tools ffmpeg libsm6 libxext6 libgl1 python3-opencv \
> >         libegl1-mesa libegl1 libosmesa6-dev patchelf \
> >     && rm -rf /var/lib/apt/lists/*
> > 
> > RUN python3 -m pip install -U pip && python3 -m pip uninstall -y byted-wandb
> > RUN python3 -m pip install --no-cache-dir \
> >     opencv-python==4.7.0.72 opencv-python-headless==4.7.0.72 pytorch_lightning \
> >     python-dotenv tqdm jupyter-client==7.4.4 numpy==1.19.5 g2p_en \
> >     einops einops-exts pillow more-itertools \
> >     braceexpand webdataset nltk scipy inflection sentencepiece open_clip_torch \
> >     black mypy pylint pytest requests tensorboard transformers datasets
> > RUN python3 -m pip install --no-cache-dir pytest-benchmark pytest-runner setuptools==57.5.0
> > RUN python3 -m pip install --no-cache-dir mmcv -f https://download.openmmlab.com/mmcv/dist/cu117/torch2.0/index.html
> > RUN python3 -m pip install pyhash
> > RUN python3 -m pip install open-flamingo
> > 
> > RUN sed -i "s/^from decimal import Decimal/from math import gcd\n&/" /usr/lib/python3.9/fractions.py
> > RUN python3 -m pip uninstall -y wandb && python3 -m pip install -U --force-reinstall wandb
> > 
> > RUN rm /usr/bin/python && ln -s /usr/bin/python3 /usr/bin/python
> > ```
> 
> Hi, why the versions of python library in the provided docker config is different from the `https://github.com/RoboFlamingo/RoboFlamingo/blob/main/requirements.txt`
> 
> Which one should I follow?

You can follow the `requirement.txt`, as we use that image as a basic image.

Ericonaldo, 2024-02-29:
> The previous errors have been solved. Now I have a new problem:
>
> Traceback (most recent call last): File "/home/x_dingj/anaconda3/envs/RoboFlamingo/lib/python3.8/threading.py", line 932, in [...]

Seems the OpenGL is not run correctly on your machine.

dingjiansw101, 2024-02-29:
What are the opengl and cuda versions that you use?

Ericonaldo, 2024-03-01:
Can refer to this [issue](https://github.com/RoboFlamingo/RoboFlamingo/issues/6)

dingjiansw101, 2024-03-11:
I followed this link to solve the previous bug of rendering with nvidia. https://sourceforge.net/p/pyopengl/bugs/258/

What is the vendor that you use, nvidia or mesa? When I tried to reproduce your results. The nvidia is only 3.944. The mesa is 4.036, which is closer to your reported 4.09, but still a little lower.

Ericonaldo, 2024-04-06:
I think it is mesa