RoboFlamingo/RoboFlamingo #20: open-loop control？

lunalulu, 2024-02-26:
[image]
like above image，it come from the paper abstract，why is open-loop control（task）？Are we not supposed to solve a closed-loop task?

Ericonaldo, 2024-02-29:
Hi, in our paper we mainly show close-loop control. 
Here by "open-loop control" we mean that RoboFlamingo has such a flexibility to do that as we have a light-weight, small policy head that can be directly deployed onto low-computation platforms with high frequency, and the large model part should be served as a cloud service run under a low frequency.

lunalulu, 2024-02-29:
> Hi, in our paper we mainly show close-loop control. Here by "open-loop control" we mean that RoboFlamingo has such a flexibility to do that, as we have a light-weight, small policy head that can be [...]

thanks very much ,
Consistent with my understanding！

closed 2024-02-29 by Ericonaldo