RoboFlamingo/RoboFlamingo #22: Window size

retsuh-bqw, 2024-03-05:
I tried to follow your instructions to reproduce the results with a *mpt-1b-redpajama-200b-dolly* model trained on the ABC->D split. With all other training parameter settings consistent, to speed up training I set the window size parameter to 4. But the final test result is merely 0.43 Avg. len (detailed resuts are 37.8/ 5.0/ 1.2/ 0/ 0 success rate for tasks 1 to 5). 

How does window size affect performance? Do you have any relevant ablation experiments?

```
torchrun --nnodes=1 --nproc_per_node=8 --master_port=6042 robot_flamingo/train/train_calvin.py \
    --report_to_wandb \
    --llm_name mpt_dolly_3b \
    --traj_cons \
    --use_gripper \
    --fusion_mode post \
    --rgb_pad 10 \
    --gripper_pad 4 \
    --precision fp32 \
    --num_epochs 5 \
    --gradient_accumulation_steps 1 \
    --batch_size_calvin 6 \
    --run_name RobotFlamingoDBG \
    --calvin_dataset ${calvin_dataset_path} \
    --lm_path ${lm_path} \
    --tokenizer_path ${tokenizer_path} \
    --openflamingo_checkpoint ${openflamingo_checkpoint} \
    --cross_attn_every_n_layers 4 \
    --dataset_resampled \
    --loss_multiplier_calvin 1.0 \
    --workers 1 \
    --lr_scheduler constant \
    --warmup_steps 5000 \
    --learning_rate 1e-4 \
    --save_every_iter 10000 \
    --from_scratch \
    --window_size 4 #> ${log_file} 2>&1
```

Ericonaldo, 2024-03-05:
Did you align the training and test window size? (Along with other setups)

retsuh-bqw, 2024-03-05:
I thought so. Since all the configurations for evaluation are read from the ckpt path. 

I also tried to set it manually in the *eval_ckpts.py* as:

```
for ckpt_name in ckpt_names:
    use_gripper = 1 if 'gripper' in ckpt_name else 0
    use_state = 1 if 'state' in ckpt_name else 0
    ckpt_path = 'roboflamingo_original/checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_4_mpt_dolly_3b_4.pth4.pth'
    os.makedirs('logs_new', exist_ok=True)
    log_file = 'logs_new/evaluate_{}.log'.format(ckpt_name.split('.')[0])
    ckpt_ix = ckpt_names.index(ckpt_name)
    print('evaluating {}/{} checkpoint'.format(ckpt_ix+1, len(ckpt_names)))
    fusion_mode = 'pre'
    if 'post' in ckpt_name:
        fusion_mode = 'post'
    if 'two_way' in ckpt_name:
        fusion_mode = 'two_way'

        
    window_size = 4
    node_num = 4

    os.system('bash robot_flamingo/pt_eval_ckpts.bash {} {} {} {} {} {} {}'.format(ckpt_path, log_file, use_gripper, use_state, fusion_mode, window_size, node_num))
```

closed 2024-03-26 by retsuh-bqw