RoboFlamingo/RoboFlamingo #26: dimmension of state,15 or 7 ???

lesjie-wen, 2024-03-14:
l have read the code and found that the dimmension of state is ambigous.
First, in BCflamingo, the state_dim is 15 (default) and l ensure that no specified state_dim when create this model.
[image]
And in action head, the state_dim is 7. Which can not match when forward passing in BCflamingo.
[image]
Can somebody help? l will appreciate that.

lesjie-wen, 2024-03-15:
Okay, l notice that all the released scripts not set the use_state True which means the model will not use state_dim. So, if l want to use the state, l need to specify the state_dim in scripts.

Ericonaldo, 2024-04-06:
It is true that we did not use any proprioceptive state for training, but you can choose either 15 or 7 for the state dim, and I think 7 means 'joints positions' or 'ee pose+gripper'.

lesjie-wen, 2024-04-28:
Ok, thanks for your reply.

closed 2024-04-28 by lesjie-wen