RoboFlamingo/RoboFlamingo #30: Why are the success rates decrease as the task index increase?

zwbx, 2024-03-31:
Hi, thanks for your work! I am seeking to understand why, within the CALVIN benchmark results, there is a noticeable trend where the success rates decline as the task index increases. Despite reviewing the CALVIN paper, I have not identified a clear rationale for this pattern. Could you provide insights.

zwbx, 2024-03-31:
By "In addition, the success rate of the subsequent tasks can be regarded as a notion of the generalizability
of the manipulation policies, since the initial state of a subsequent task highly relies on the ending
state of its former task.", the view you point out in the paper, you mean the diverse initial state brings the challenge. My question is that:

1. What does the initial state specifically indicate? the visual observation or the joint angles of the arm?
2. Why are the initial state of later tasks more diverse than the former ones? is this due to in the training set, there is no failure ending state as a initial state for the later task, while at inference stage, there exists error accumulation, causing the diverse state of the later task.

mbreuss, 2024-04-05:
Hi,

the task rate can only decrease, because the model only gets the second task after completing the first one during a rollout. If the model completes the first task in 90% during 1000 rollouts, the second task success rate is still meassured in relative to 1000 rollouts, while the model will only received the second tasks 900 times. As a consequence the task rate will decrease with higher task number.
Hope this helps.

zwbx, 2024-04-05:
Thanks, I would not have thought the reason is such straightforward.

closed 2024-04-05 by Ericonaldo