RoboFlamingo/RoboFlamingo #31: Evaluation with comes up with strange results

Qianrj0735, 2024-04-11:
Hi, It is a impressing work, however, when I tried to evaluate with the config

- mpt-1b-redpajama-200b-dolly
- Loading origin flamingo checkpoint from  OpenFlamingo-3B-vitl-mpt1b-langinstruct/checkpoint.pt
- Loading robot-flamingo checkpoint from RoboFlamingo/checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth

The performance in Calvin seems weird, please view the gif
[image]
[image]
[image]
please feel free to ask me for more evaluation details

Ericonaldo, 2024-04-12:
What evaluation scripts are you using?

Qianrj0735, 2024-04-14:
python robot_flamingo/eval/eval_calvin.py     --precision fp32     --use_gripper     --window_size 12     --fusion_mode post     --run_name RobotFlamingoDBG     --calvin_dataset my/path/to/calvin   --evaluate_from_checkpoint checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth     --calvin_conf_path calvin/calvin_models/conf     --workers 1

and I noticed that in the original code --lm_path  --tokenizer_path  --cross_attn_every_n_layers is defined by mpt_dict in factory.py, and their value is respectively:

lm_path = mpt-1b-redpajama-200b-dolly
tokenizer_path = mpt-1b-redpajama-200b-dolly
cross_attn_every_n_layers = 1
openflamingo_checkpoint = OpenFlamingo-3B-vitl-mpt1b-langinstruct/checkpoint.pt

Qianrj0735, 2024-04-14:
window size: 12
1024 2048
MosaicGPTConfig {
  "_name_or_path": "mpt-1b-redpajama-200b-dolly",
  "alibi": true,
  "alibi_bias_max": 8,
  "architectures": [
    "MosaicGPT"
  ],
  "attn_clip_qkv": null,
  "attn_impl": "torch",
  "attn_pdrop": 0,
  "attn_qk_ln": true,
  "attn_uses_sequence_id": false,
  "auto_map": {
    "AutoConfig": "configuration_mosaic_gpt.MosaicGPTConfig",
    "AutoModelForCausalLM": "mosaic_gpt.MosaicGPT"
  },
  "d_model": 2048,
  "emb_init_std": null,
  "emb_init_uniform_lim": null,
  "emb_pdrop": 0,
  "embedding_fraction": 1.0,
  "fan_mode": "fan_in",
  "init_device": "cpu",
  "init_div_is_residual": true,
  "init_gain": 0,
  "init_nonlinearity": "relu",
  "init_std": 0.02,
  "logit_scale": null,
  "low_precision_layernorm": true,
  "max_seq_len": 2048,
  "mlp_ratio": 4,
  "model_type": "mosaic_gpt",
  "n_heads": 16,
  "n_layers": 24,
  "no_bias": true,
  "param_init_fn": "kaiming_normal_",
  "prefix_lm": false,
  "resid_pdrop": 0,
  "softmax_scale": null,
  "tokenizer_name": "EleutherAI/gpt-neox-20b",
  "torch_dtype": "float32",
  "transformers_version": "4.33.1",
  "use_cache": false,
  "verbose": 0,
  "vocab_size": 50280
}

----------------------------------------------------------------------------------------------------
transformer.blocks
Flamingo model initialized with 1088187703 trainable parameters
Loading origin flamingo checkpoint from  OpenFlamingo-3B-vitl-mpt1b-langinstruct/checkpoint.pt
Loading robot-flamingo checkpoint from checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth

**output from the script**

Ericonaldo, 2024-04-14:
You can refer to `https://github.com/RoboFlamingo/RoboFlamingo/blob/main/robot_flamingo/pt_eval_ckpts.bash`, and I think the key problem is `--cross_attn_every_n_layers 4`

Qianrj0735, 2024-04-15:
thanks! I tried with --cross_attn_every_n_layers 4 but it's not working, I didn't use pt_eval_ckpts.bash since I didn't have a multi-gpu environment, but now I have a server with several gpus, I would try with that bash

Ericonaldo, 2024-04-15:
There is another problem. As you can see the name `checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth`, this ckpt is trained on `mpt_dolly_3b`, but you load `mpt_dolly_1b`, can you try this?

SatireY, 2024-04-17:
> There is another problem. As you can see the name `checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth`, this ckpt is trained on `mpt_dolly_3b`, but you load `mpt_dolly_1b`, can you try `mpt_dolly_3b`?
I have a question about this. Shouldn't the LLM ckpt corresponding to MPT-3b-IFT mentioned in the paper be "mpt_dolly_1b" instead of "mpt_dolly_3b"?

Ericonaldo, 2024-04-17:
no, MPT-3b-IFT is MPT-3b with instruction fine-tuning

SatireY, 2024-04-17:
> no, MPT-3b-IFT is MPT-3b with instruction fine-tuning

I have some confusion. I see in the readme document of the project that the longest model of "AvgLen" is model 3B, and the corresponding Language model is "anas-awadalla/mpt-1b-redpajama-200b-dolly". This is also the best result in the paper (AvgLen=4.09), I do not see roboFlamingo using the mpt-3b model in the paper. Also roboFlamingo's base model OpenFlamingo doesn't use the mpt-3b model as a language model, I only see the 4B OpenFlamingo model for 'togethercomputer/RedPajama-INCITE-Base-3B-v1',Why is that?
[image]

Ericonaldo, 2024-04-17:
Sorry, you are right, my mistake, `checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth` corresponds to `mpt_dolly_1b`

SatireY, 2024-04-17:
> Sorry, you are right, my mistake, `checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth` corresponds to the language model `mpt_dolly_1b`

Ok, I understand. Thank you for your quick reply

Ericonaldo, 2024-04-18:
@SatireY can use https://github.com/RoboFlamingo/RoboFlamingo/blob/main/eval_ckpts.py for evaluation

SatireY, 2024-04-18:
Thank you for reminding me, I will try. Another thing that makes me confused is about the setting of the parameter `--cross_attn_every_n_layers`. As you can see in the picture, the value of 'Xattn interval' of model 3B should be 1, but it is indeed 4 when used in scripts.Why is that?
[image]

Ericonaldo, 2024-04-18:
Thanks for reminding me. You are right. The arg is replaced by the one defined in the dict, so `--cross_attn_every_n_layers 4` does not make any difference. I apologize for your confusion. As for your strange results, may be you can try that eval script first to see if there is any difference.

SatireY, 2024-04-18:
> Thanks for reminding me. You are right. The arg is replaced by the one defined in the dict, so `--cross_attn_every_n_layers 4` does not make any difference. I apologize for your confusion. As for your strange results, you can try that eval script first to see if there is any difference.

Ok, I will try to run, if there is any confusion may need your help, thank you very much for your quick reply!

SatireY, 2024-04-23:
> thanks! I tried with --cross_attn_every_n_layers 4 but it's not working, I didn't use pt_eval_ckpts.bash since I didn't have a multi-gpu environment, but now I have a server with several gpus, I would try with that bash

I get an error when I try to execute the validation script:

```
AttributeError: Error instantiating 'calvin_env.envs.play_table_env.PlayTableSimEnv' : Error instantiating
'calvin_env.camera.tactile_sensor.TactileSensor' : module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by
itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use
`np.float64` here.
` ` `
Can you share with me the environment in which you successfully run eval?, thank you for your help.

zimu-bina, 2024-12-02:
> Hi, It is a impressing work, however, when I tried to evaluate with the config
> 
> * mpt-1b-redpajama-200b-dolly
> * Loading origin flamingo checkpoint from  OpenFlamingo-3B-vitl-mpt1b-langinstruct/checkpoint.pt
> * Loading robot-flamingo checkpoint from RoboFlamingo/checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_dolly_3b_2.pth
> 
> The performance in Calvin seems weird, please view the gif ![2-0-lift_pink_block_slider-fail](https://private-user-images.githubusercontent.com/40055876/321536162-4c5a7f44-9442-4a6c-87ac-ff7f38f86147.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDA1ODIsIm5iZiI6MTczMzEwMDI4MiwicGF0aCI6Ii80MDA1NTg3Ni8zMjE1MzYxNjItNGM1YTdmNDQtOTQ0Mi00YTZjLTg3YWMtZmY3ZjM4Zjg2MTQ3LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAyVDAwNDQ0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM0ZDUyM2NlMGVhNjlkYmFiYTMwMzdlZDhhOTczNDNmMzRjZTBhM2M0MWE3MDFiZDg0N2Q0Y2FkMTA2NzU0NDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.icNK4YneaZwo4sqqnoLGV2txzMXzu5smzuYx2sx1Yx0) [ ![2-0-lift_pink_block_slider-fail](https://private-user-images.githubusercontent.com/40055876/321536162-4c5a7f44-9442-4a6c-87ac-ff7f38f86147.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDA1ODIsIm5iZiI6MTczMzEwMDI4MiwicGF0aCI6Ii80MDA1NTg3Ni8zMjE1MzYxNjItNGM1YTdmNDQtOTQ0Mi00YTZjLTg3YWMtZmY3ZjM4Zjg2MTQ3LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAyVDAwNDQ0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM0ZDUyM2NlMGVhNjlkYmFiYTMwMzdlZDhhOTczNDNmMzRjZTBhM2M0MWE3MDFiZDg0N2Q0Y2FkMTA2NzU0NDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.icNK4YneaZwo4sqqnoLGV2txzMXzu5smzuYx2sx1Yx0) ](https://private-user-images.githubusercontent.com/40055876/321536162-4c5a7f44-9442-4a6c-87ac-ff7f38f86147.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDA1ODIsIm5iZiI6MTczMzEwMDI4MiwicGF0aCI6Ii80MDA1NTg3Ni8zMjE1MzYxNjItNGM1YTdmNDQtOTQ0Mi00YTZjLTg3YWMtZmY3ZjM4Zjg2MTQ3LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAyVDAwNDQ0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM0ZDUyM2NlMGVhNjlkYmFiYTMwMzdlZDhhOTczNDNmMzRjZTBhM2M0MWE3MDFiZDg0N2Q0Y2FkMTA2NzU0NDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.icNK4YneaZwo4sqqnoLGV2txzMXzu5smzuYx2sx1Yx0) [ ](https://private-user-images.githubusercontent.com/40055876/321536162-4c5a7f44-9442-4a6c-87ac-ff7f38f86147.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDA1ODIsIm5iZiI6MTczMzEwMDI4MiwicGF0aCI6Ii80MDA1NTg3Ni8zMjE1MzYxNjItNGM1YTdmNDQtOTQ0Mi00YTZjLTg3YWMtZmY3ZjM4Zjg2MTQ3LmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAyVDAwNDQ0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM0ZDUyM2NlMGVhNjlkYmFiYTMwMzdlZDhhOTczNDNmMzRjZTBhM2M0MWE3MDFiZDg0N2Q0Y2FkMTA2NzU0NDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.icNK4YneaZwo4sqqnoLGV2txzMXzu5smzuYx2sx1Yx0) ![1-0-turn_off_led-fail](https://private-user-images.githubusercontent.com/40055876/321536173-a2aff43b-6b53-4011-ba56-88f50bdc38bb.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDA1ODIsIm5iZiI6MTczMzEwMDI4MiwicGF0aCI6Ii80MDA1NTg3Ni8zMjE1MzYxNzMtYTJhZmY0M2ItNmI1My00MDExLWJhNTYtODhmNTBiZGMzOGJiLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAyVDAwNDQ0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTczODdkNDliMWE5NDdjMzE1MDcxY2RkYzBhMzkyNjRhZTYyZWNmYmJhYmU5MWZmZjUzMzZmMGY0ZTg4MjcyNmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gg5Q1JjVRY3OMwWgExvbo4C09GBnk7mhVTXFIa5F8MY) [ ![1-0-turn_off_led-fail](https://private-user-images.githubusercontent.com/40055876/321536173-a2aff43b-6b53-4011-ba56-88f50bdc38bb.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDA1ODIsIm5iZiI6MTczMzEwMDI4MiwicGF0aCI6Ii80MDA1NTg3Ni8zMjE1MzYxNzMtYTJhZmY0M2ItNmI1My00MDExLWJhNTYtODhmNTBiZGMzOGJiLmdpZj9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAyVDAwNDQ0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTczODdkNDliMWE5NDdjMzE1MDcxY2RkYzBhMzkyNjRhZTYyZWNmYmJhYmU5MWZmZjUzMzZmMGY0ZTg4MjcyNmImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gg5Q1JjVRY3OMwWgExvbo4C09GBnk7mhVTXFIa5F8MY) ](https://private-user-images.githubusercontent.com/40055876/321536173-a2aff43b-6b53-4011-ba56-88f50bdc38bb.gif?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMxMDA1ODIsIm5iZiI6MTczMzE [... cut ...] ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjAyVDAwNDQ0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ0OTA3NGI4ZDAxNDVkODgwMjBlZTQ1NjhiODVmN2ExMzExY2UyYmU0YzI1MGNhMTgwZWJjNzA4YmE1ZTdiYTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.mgMBRWrlfkmcjJygjWwZX53ThE8T6NwdUV0Jxglwx8w) please feel free to ask me for more evaluation details

Hello, Professor. I have the same question as you. Have you solved it?