RoboFlamingo/RoboFlamingo #32: about training data？

lunalulu, 2024-04-11:
If I understand correctly, all the data we need actually does not require the internal and external calibration parameters of the camera, right?

thanks very much

Ericonaldo, 2024-04-12:
The public code is for training on simulated benchmark, so yes. If you need to deploy, you would need calibration.

lunalulu, 2024-04-12:
> The public code is for training on simulated benchmark, so yes. If you need to deploy it onto real robots, you would need calibration.

I understand that we are an end-to-end network, from observation to action, so why is calibration needed on the actual robotic arm?
thanks for your reply~

Ericonaldo, 2024-04-12:
You will need to calibrate like the camera position and latency. You will know that when you conduct real robot experiments.

oym1994, 2024-04-12:
> You will need to calibrate like the camera position and latency. You will know that when you conduct real robot experiments.

Hi, we also face this problem. In theory, if we can guarantee the camera position and latency(something else) are same between training data collection and deployment period, there will need no calibration. I think you and your team must have met and solved much engineer problem. So would you mind introducing some practical experience for that? Many thanks.

Ericonaldo, 2024-04-12:
It is case by case.

closed 2024-04-12 by Ericonaldo