RoboFlamingo/RoboFlamingo #39: Failed on all eval tasks

FanZhang91, 2024-05-10:
Language model:  anas-awadalla/mpt-1b-redpajama-200b
Vision encode:  openai CLIP ViT-L/14
finetune model:  pretrained model  checkpoint_gripper_post_hist_1_aug_10_4_traj_cons_ws_12_mpt_3b_4.pth(download from Hugging Face website)
eval_data:  eval_sequences.json

run eval_calvin.py to evaluate model, all task failed. Just like the following gif:
[image]
[image]
[image]
[image]
[image]

Can you help me solve this problem?

Ericonaldo, 2024-05-17:
Hi, the default models are trained on ABCD->D dataset, and the ones in the directory are trained on ABC->D dataset (see CALVIN paper for details.)

Ericonaldo, 2024-05-17:
It is hard to say the reason. Do you make sure all parameters are the same with the downloaded model?

FanZhang91, 2024-05-17:
> It is hard to say the reason. Do you make sure all parameters are the same with the downloaded model?

I use CALVIN debug dataset. Except for the parameter “cross_attn_every_n_layers”, everything else is the same。The paramer “cross_attn_every_n_layers” used in training is from factory.py file(pretrained model use cross_attn_every_n_layers=1, while i use cross_attn_every_n_layers=4). The training script is as follows:
python train_calvin.py --report_to_wandb --llm_name mpt_3b --traj_cons --use_gripper --fusion_mode post --rgb_pad 10 --gripper_pad 4 --precision fp32 --num_epochs 1000 --gradient_accumulation_steps 10 --batch_size_calvin 6 --run_name RobotFlamingoDBG --calvin_dataset /datasets/calvin_debug_dataset --cross_attn_every_n_layers 4 --dataset_resampled --loss_multiplier_calvin 10.0 --workers 1 --lr_scheduler const --warmup_steps 5000 --learning_rate 1e-4 --save_every_iter 10000 --window_size 4。

zimu-bina, 2024-12-02:
Hello, Professor. I have the same question as you. Have you solved it?