RoboFlamingo/RoboFlamingo #40: The version of used PyTorch

hnuzhy, 2024-05-13:
It seems that this project [RoboFlamingo](https://github.com/RoboFlamingo/RoboFlamingo), the dataset API [calvin_agent](https://github.com/mees/calvin) and the 3rd library [open_flamingo](https://github.com/mlfoundations/open_flamingo) have used three different versions (`torch==1.12.1+cu113`, `torch==1.13.1`, and `torch==2.0.1`) of the PyTorch. So what should I do to choose a suitable one?

Ericonaldo, 2024-05-17:
Hi I think the latest version should work fine.

hnuzhy, 2024-05-17:
> Hi I think the latest version should work fine.

Ok. I have already launched my running script. However, I can only use one A100 GPU, with unfixed bugs when using multi-GPUs. Hope the final training results will be fine.

closed 2024-05-17 by hnuzhy